Uzebox/uzebox #21: gconvert needs a Make install rule

felipesanches, 2015-06-13:


Uzebox, 2015-06-20:
Why is that? Only gconvert?

felipesanches, 2016-08-07:
well... we need an install rule for the tools, not only gconvert.

nicksen782, 2019-01-22:
Do you mean a Makefile that would compile all the tools in the tools directory instead of individual compiles?

felipesanches, 2019-01-22:
**install**

felipesanches, 2019-01-22:
it's been more than 3 years... I would have to look carefully to remember what I originally meant.

But I said "Make install", so it seems the issue was with installing, not compiling.

weber21w, 2023-07-07:
All tools seem to build in the current codebase? Shall we close?

artcfox, 2023-07-07:
I think the OP might have meant if you go to the gconvert directory and type make, it will make it in its source directory, rather than make it in the bin/ directory. The solution is to make it from the very top level directory. I think it's safe to close this.

closed 2023-07-10 by weber21w