Uzebox/uzebox #52: Uzem case sensitivity

benpocalypse, 2015-12-31:
Under *nix systems, the Uzebox emulator converts input filenames to lowercase, and then won't open the corresponding roms if they're not in lowercase. If you manually convert the .hex or .uze filename generated for the demos to lowercase, everything works fine, but as it's generated, they're mixed case.

Example:
./uzem SuperMarioDemo.hex
Result:
Error: cannot load hex image 'supermariodemo.hex'

Example 2:
./uzem supermariodemo.hex
Result:
All good.

I would issue a patch/pull request, but I don't have the time at the moment. If I can see where the conversion is made in the uzem source, I'll update my issue.
Thanks,
and Happy New Years!

miracoli, 2016-01-03:
I issued a pull request for this.
Happy new year!

felipesanches, 2016-08-07:
This seems to be fixed. I cannot replicate the issue on my local master branch build. Please close the issue.

felipesanches, 2016-08-17:
@uze6666 could you please close this issue?

closed 2016-10-23 by uze6666