Uzebox/uzebox #56: Uzem is hard to use on hidpi displays

nebososo, 2016-02-28:
I'm using a 3200x1800 screen on linux. Most programs will scale 2.5x so it feels like 1280x720 (GTK and qt do this), but SDL/uzem won't. Uzem's windows is really tiny and I have to bring to bring the screen closer to see something.
I believe an option for scaling would solve the problem. For example: uzem -s 2.5 game.uze
I took a quick look at the code and it doesn't seem trivial, but someone who knows the code better might be able to give a more educated opinion.

uze6666, 2016-02-28:
Have you tried building uzem from the uzem140 branch? There was a ton of improvements made on speed and video display scaling last year that I will merge it back to the trunk soon. Would be good to get a chance to see if it resolves your issue.

nebososo, 2016-02-28:
Nice. I didn't even see that branch. It's much, much better, it sure fixes my issue, great stuff.

uze6666, 2016-02-29:
Great then! The refresh rate is also improved. Fore some reasons, it was 30 hz before. Now it's been fixed back to 60hz so scrolling and animations are much smoother.

closed 2016-02-29 by uze6666