Uzebox/uzebox #59: Uzem as libretro core

t-paul, 2016-05-08:
I've been porting Uzem to be able to run as [libretro](http://www.libretro.com/) core. So before sending big pull requests I'd like to get some feedback first. Is that something of interest at all?

*What is libretro?*
It an API that can be implemented by emulator cores or games providing a cross platform abstraction layer such that the emulator does not need to handle all the system specific code themself.

*Why is that useful?*
If the emulator is compiled as core with the libretro API, it can be loaded into any frontend that provides support for this. The best known frontend application is RetroArch (also prominently featured in [Lakka](http://www.lakka.tv/) and the HackADay project [GameGirl](https://hackaday.io/project/10207-gamegirl-the-retro-console-done-right)) but there are others in the works, like the RetroPlayer that should soon appear in Kodi.

RetroArch provides a vast amount of extended features like video shaders to improve the screen output or frame based rewind to go back in emulation time by just pressing a button.

*What's the current status?*
I've separated most of the SDL code and moved it to drivers for audio, input and video. At compile time, it's then possible to link either the SDL drivers and get the usual standalone emulator binary or link against the libretro drivers and produce the shared library core.

*How does it look like?
See https://youtu.be/ky8KJLScOpI for a short screencast running LodeRunner and Arkanoid with a simple scanline video shader enabled.

*Where is it?*
Introduce driver interface: Branch [`driver interface`](https://github.com/t-paul/uzebox/tree/driver-interface)
Add libretro drivers: Branch [`libretro`](https://github.com/t-paul/uzebox/tree/libretro) based on the `driver interface` branch

NerosTie, 2017-05-09:
Any news about this project? I'm very interested!

weber21w, 2017-07-03:
Also interested in this, any idea on the minimum system specs to run full speed?

Jubatian, 2017-07-03:
Now Debian contains RetroArch in Stretch, I just upgraded before weekend. This makes it a little more convenient (at least for me) to do something about it (no need to install "third party" stuff and-or mess around with library paths to get things going). For Uzem, likely no, but I guess I will try this for CUzeBox as I originally intended to.

t-paul, 2017-07-03:
@weber21w It runs at about half speed on a Raspberry Pi 3, should be fine with any decent PC.

It will be interesting to see how it works in Gnome-Games or the Kodi Retroplayer.

uze6666, 2019-01-22:
Anyone interested in porting this? Otherwise I will close this issue.

t-paul, 2019-01-22:
Huh? It's ported and part of Lakka for years now (not sure of the latest state though). The issue was meant to get feedback if there's a way to upstream things. By now the libretro team seems to have created an additional fork which I did not follow.

paradadf, 2019-07-12:
@t-paul would you mind updating your [uzebo's libretro branch](https://github.com/t-paul/uzebox/tree/libretro) and PR your changes so it gets merged upstream?

Jubatian, 2019-07-15:
A problem here is that Uzem is more or less discontinued, nowadays rather CUzeBox is used as "official" Uzebox emulator ( https://github.com/Jubatian/cuzebox ). The reason behind abandoning it was mostly issues in its design which prevented moving the codebase forwards in some regards. The newer CUzeBox is faster (introducing such performance improvements which the structure of Uzem didn't permit), and notably has SPI RAM and SD card write support, both of which are used by a few newer games (the latter is the more important).

So it might be beneficial to rather pick up CUzeBox for a LibRetro core, if for nothing else, the improved performance (maybe permitting it running on things it didn't quite run on before).

t-paul, 2019-07-21:
Nice, I've not seen CUzeBox before. The libretro core of Uzem did run at about 50% speed on a raspi3. Maybe raspi4 + CUzeBox would just make it?

Not sure how it looks with spare time for this, but maybe I can have a look at CUzeBox instead. I have not looked at the latest state, but with Gnome-Games and Kodi implementing libretro support it would be nice to see Uzebox supported there too.

Jubatian, 2019-07-21:
It might do.

One particular thing to look at could be, ridiculous as it sounds, the implementation's performance in shovelling pixels from memory to the display. In the web emulator running at 60FPS in high quality mode, getting the pixels on the display (within Emscripten's SDL) actually takes more CPU than running the AVR emulation. In native PC builds, this is much less ridiculous, but still substantial. So coming up with a good solution to this might be a game changer, this is also easier to do with CUzeBox than Uzem as the display is well separated.

I might look at LibRetro myself some time later, although maybe much later, being rather interested in other sorts of things these times.

weber21w, 2023-07-07:
It sounds like no one is porting the new official CUzeBox emulator over, and Uzem is long outdated. Perhaps at a later date this will be port will pursued, but that should be a separate non-Uzem discussion as this has gone stale.

closed 2023-07-07 by weber21w