Uzebox/uzebox #61: Megatris: Should not allow infinite rotation per tetramino

felipesanches, 2016-08-07:
During gameplay, if the player keeps quickly rotating the tetramino, then the game will never give you a new tetramino. This can be used as a way of cheating in the game.

I suggest that there should be a timeout for each tetramino that starts counting once the tetramino touches other blocks in the playfield.

uze6666, 2016-08-16:
Hi Felipe, this is a feature, not a bug. The reference was TetrisDS gameplay which allows it. That said, there could be an option in the oprtions menu to turn it off.

weber21w, 2017-07-03:
Theoretically in a versus match, it would be a terrible tactic as they build up much garbage against you while you screw around with 1 piece. Now if one was clearly going to lose, they might rotate indefinitely to not allow the opponents garbage to come and cause a loss. It seems to me this is one of those things that is easily fixed with a punch to the guy sitting next to you, otherwise are there ideas on the easiest way to implement this in code avoiding the justified physical punch? If so lets implement it.

closed 2023-07-07 by weber21w