Uzebox/uzebox #63: Mode 3 scrolling bug

weber21w, 2016-10-23:
The issue was discovered and solved by Artcfox, want to make sure we remember to add[ this fix](http://uzebox.org/forums/viewtopic.php?f=3&t=2400) in the next version.

artcfox, 2017-07-03:
The fix I applied broke mode 3 scrolling with 24 tiles tall (different wrap-around trick) so I reverted it. This bug still stands, we just need to find a better way to fix it, or nix the 24 tile tall scrolling mode. Does anyone use that?

weber21w, 2017-07-04:
Note that I know of. Would #define be in too many places to make both solutions exist?

Jubatian, 2017-08-26:
Does the recent Mode 3 blitter boost fix this bug?

Jubatian, 2017-08-27:
As far as my testing shows, the bug seems to be gone. ( http://uzebox.org/forums/viewtopic.php?f=3&t=9389&start=20#p27206 ). If you can confirm, then this could be closed.

uze6666, 2019-01-22:
Since we haven't hear anymore about this issue, I'm closing it.

closed 2019-01-22 by uze6666