Uzebox/uzebox #66: Sound effect distortions caused by kernel bugs

uze6666, 2016-11-23:
Excerpt from forum post: [http://uzebox.org/forums/viewtopic.php?f=3&t=2404](http://uzebox.org/forums/viewtopic.php?f=3&t=2404)

It seems like Artcfox then was near hitting the VSync "barrier", as we talked about it a few weeks before where he realized he had lots of cycles still remaining before the next frame. Currently Bugz is designed to work with so little CPU resources that it never hits the VSync barrier. So I believe what he experienced then relates to these, and maybe he also mistook this to be running out of VBlank time (which isn't true as there are plenty of cycles after VSync as well, even if there are many Mode 3 sprites to blit).

For the second part (possible clicks for effect when they retrigger) I think a useful fix would be making line 929 & 930 conditional: note 0xFF could indicate that those mixer updates doesn't need to be performed. This would let both music keeping working and effects not attempting to set up improper parameters for a frame (if TriggerFX is called just after VSync). Of course TriggerFx also needs to call TriggerCommon with note 0xFF.

For the first part my best bet would be adding something like the CONTROLLERS_VSYNC_READ flag for controllers, so allowing the user to start music processing instead of the default behaviour if the user needs this. This case the WaitVSync routine would also need to call ProcessMusic (otherwise for example the intro logo would become incapable to play the "bling", neither could you use WaitVSync with any parameter other than 1 if you wanted music to keep playing).

Fixing the race condition itself would be messy. It is not a matter of a volatile missing or being there, the track is a big array, and the problem is that you can't do an atomic update on that array, while you neither can block interrupts (since it can only smooth out 5 cycles of delay). A possibility could be something like I did for the sprite engine of Mode 74, but that's assembly. There I carefully ordered tasks so no matter where the processing (of RAM tile management and blitting) is cut, the result will remain consistent (only parts of sprites missing for which there wasn't time to blit). In C it is impossible to do this since the compiler is allowed to reorder stuff. Or you would have to make the whole track array volatile in order to be able to do it, but that could cripple performance (in track processing). So I don't think actually fixing this with leaving music processing in VSync is possible without drastic amount of changes (possibly moving lots of stuff to assembler).
Post details
Re: Sound effect distortions caused by kernel bugs
Post by D3thAdd3r » Fri Nov 18, 2016 3:53 pm

Some parts of this sound similar to Artcfox's discovery, but a bit different too. There was a race condition in the way the kernel triggered effects, and I am not sure it was ever uploaded to Github. Perhaps this could resolve the issue.

Otherwise I think the only 100% bullet proof solution would require 2 volatile variables to ensure a "lock" between interrupts..or something.
Post details
Sound effect distortions caused by kernel bugs
Post by Jubatian » Wed Nov 16, 2016 3:35 pm

Tracking down a problem in my UCC game I figured out several problems in the way the kernel produces sound effects.

Just describing these for now, for anyone interested, or in case you ran into something related.

The problems are hazards between the main program and interrupt, so things which occur rarely, and when they do, they are more or less annoying. There doesn't seem to be anything which could actually break a program, these are only partial or complete destructions of a played sound effect (so the effects could only last so long the effect is played).

The most radical distortion happens when TriggerFx is interrupted by the VSync, and in that, ProcessMusic is called. ProcessMusic is responsible for executing the sound patches. You can imagine what happens: TriggerFx tries to set up a patch to be played, it is halfway through, then ProcessMusic tries to step it. Bang. As far as I see this is not very harmful (as in like barfing random data all over the memory), possibly only to the ears until the crippled effect lasts.

You may hack this problem away in two ways if you are affected by it, both involves removing ProcessMusic from the VSync tasks. You may either add it after the video mode's return, or if you run your game in sync with the video (at 60Hz, such as my case with Mode 74 and its reset concept), you may also add it to the user code (exposing it on the kernel's interface).

But if you do either, you will face another problem which will get more pronounced. In the normal kernel this problem may well go unnoticed if your game triggers effects before the VSync would happen, if your effects aren't radically affected by the pitch and waveform, or if the effects are rare enough. The problem is in TriggerFx, that it calls TriggerCommon with an actual note (80), which TriggerCommon sets up for the mixer along with waveform 0, which then immediately starts playing (a sample every line). Until ProcessMusic is called, the patch is not processed, so if your effect has a fixed pitch (within the patch), that pitch won't be in effect until then. If calling the effect is past VSync, this means that for around 17 milliseconds the effect will play with improper pitch and waveform at whatever volume was last set up for the channel (this means it becomes very audible when retriggering as then there is a nonzero volume on the channel). If you apply a hack like above, this problem will become permanent (since it means all TriggerFx calls will likely happen right after ProcessMusic). The audible results are more or less annoying clicks on every retrigger.

This can be hacked around by modifying firing effects so they don't touch the mixer (as the patch will properly set the mixer up anyway on the next ProcessMusic call). In TriggerCommon accessing the mixer is necessary to allow playing music (through TriggerNote).

So if anyone ever experienced "weird" problems related to effect playback, I think these are the major causes.

Jubatian, 2017-08-26:
Mode 3 now got an option to remove ProcessSprites from VSync (so you can call it where you need to). In Flight of a Dragon I have a similar feature implemented for sound which fixes this bug. I think maybe a compile time flag could be introduced for this too, to remove ProcessMusic from VSync too, so you can time it yourself. Does this sound OK?

uze6666, 2019-01-22:
@Jubatian: Sounds good. Let's close this one.

closed 2019-01-22 by uze6666