Uzebox/uzebox #68: Extra space between joypad1_status_lo and joypad1_status_lo

bardes, 2017-01-06:
I've been working on a hand optimized function to read the controllers and noticed that there seems to be an [extra byte](https://github.com/Uzebox/uzebox/blob/master/kernel/uzeboxVideoEngineCore.s#L113-L121) between the high and low portions of both joypad statuses. I'm not sure if it's intentional or not, but I'm opening an issue just in case...

Jubatian, 2017-01-06:
That part is a little messed up, but those extra bytes are necessary and are used, they are needed by the SNES mouse (See ReadJoypadExt and ReadJoypad above it, combined they use all bytes). It could be fixed, though so that if the user doesn't need the SNES mouse, 2 bytes less would be allocated.

bardes, 2017-01-06:
Oh, didn't know about that... Thanks for the quick answer!

closed 2017-01-06 by bardes