Uzebox/uzebox #98: ControllerTester doesn't work (in emulators)

danboid, 2020-06-21:
I have tried running the ControllerTester demo with both cuzebox and uzem under Ubuntu 18.04 but could not get it to register input from the emulated keypad joystick.

I don't suppose this demo only works on real hardware does it?

Jubatian, 2020-06-21:
Had seen this before, never really looked into the why. The Controller Tester works on the real hardware (at least for SNES controllers it does). As of now this seems to be the only known broken application on the emulators (CUzeBox follows the mechanism Uzem used for emulating the controllers, so the problem causing the difference is likely the same in both).

danboid, 2020-06-21:
Thanks for the explanation.

uze6666, 2023-03-03:
The tool was meant for the real hardware. Following Jubatian answer, I will closing this issue.

closed 2023-03-03 by uze6666