Uzebox/uzebox #99: Add build instructions to master

KrzysztofJeszke, 2020-08-19:
I cannot get to compile the uzebox stuff without instructions on windows, and that would be really useful to have some. As on MSYS2 it just doesn't want to build.

Jubatian, 2020-08-19:
Could you provide some output? (I assume you went for it, tried, and it failed on something) However just to note, there are some code in it not quite up to standards :p , currently actually we are in the process of trying to wrestle it back into operation on a recent version of avr-gcc (10.1.0).

KrzysztofJeszke, 2020-08-19:
`Makefile:228: *** Unsupported platform!. Stop.`
That basically shows up when making in uzebox directory and when making tools.
MSYS2 is basically MinGW so i don't see why.

Jubatian, 2020-08-30:
Sorry for the late reply, as I mentioned, we were working on getting the repo compiling on recent versions of avr-gcc, now it should at least compile.

Could you work it around in the end? By the single line you posted, I could suspect the problem being that your setup tried to compile the games (for 8-bit AVR) with a native (PC, Windows) compiler. If the problem still stands, could you post a longer section of the output, which shows what was being compiled when you received that error?

(Unfortunately I am somewhat in the dark as I have no Windows, so can't reproduce it myself or document it since I don't know how it would look like there)

KrzysztofJeszke, 2020-08-30:
I meant that the tools didn't want to build, and by so I couldn't get working.

Jubatian, 2020-08-31:
Could you post a longer section of the output? Would like to see what exactly fails with that error as there are parts in the repo which are compiled by different compilers (so for example whether an actual game fails of for example the build of packrom, which is a tool used for packaging them, running on the host).

uze6666, 2023-03-03:
Wiki was updated with working toolchain on Windows.

closed 2023-03-03 by uze6666