Uzebox/uzebox #103: uzem build fails under EOS/Arch Linux

danboid, 2022-08-20:
```
$ make
make -C tools/uzem  DEST_DIR=/home/dan/src/uzebox/bin/
make debug
=================================
Building Debug...
Platform: Unix-LINUX
=================================
g++ -c uzem.cpp -o Debug/uzem.o -I/usr/include/SDL2 -D_REENTRANT -DLINUX -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzem.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
In file included from avr8.h:33,
                 from uzem.cpp:27:
gdbserver.h:118:58: error: reference to ‘byte’ is ambiguous
  118 |         void avr_core_flash_write_hi8(unsigned int addr, byte val) ;
      |                                                          ^~~~
In file included from /usr/include/c++/12.1.1/bits/stl_algobase.h:61,
                 from /usr/include/c++/12.1.1/vector:60,
                 from avr8.h:29:
/usr/include/c++/12.1.1/bits/cpp_type_traits.h:406:30: note: candidates are: ‘enum class std::byte’
  406 |   enum class byte : unsigned char;
      |                              ^~~~
gdbserver.h:57:17: note:                 ‘typedef uint8_t byte’
   57 | typedef uint8_t byte;
      |                 ^~~~
gdbserver.h:118:58: error: ‘byte’ has not been declared
  118 |         void avr_core_flash_write_hi8(unsigned int addr, byte val) ;
      |                                                          ^~~~
gdbserver.h:119:58: error: reference to ‘byte’ is ambiguous
  119 |         void avr_core_flash_write_lo8(unsigned int addr, byte val) ;
      |                                                          ^~~~
/usr/include/c++/12.1.1/bits/cpp_type_traits.h:406:30: note: candidates are: ‘enum class std::byte’
  406 |   enum class byte : unsigned char;
      |                              ^~~~
gdbserver.h:57:17: note:                 ‘typedef uint8_t byte’
   57 | typedef uint8_t byte;
      |                 ^~~~
gdbserver.h:119:58: error: ‘byte’ has not been declared
  119 |         void avr_core_flash_write_lo8(unsigned int addr, byte val) ;
      |                                                          ^~~~
make[2]: *** [Makefile:283: Debug/uzem.o] Error 1
make[1]: *** [Makefile:264: all] Error 2
make: *** [Makefile:130: tools/uzem] Error 2
```

danboid, 2022-09-04:
I have the same problem under Debian Testing with gcc 12.2.0

danboid, 2022-09-04:
uzem can still be successfully built under Debian testing by disabling GDB support:

```
cd uzebox/tools/uzem 
NOGDB=1 make release
```

I have tried installing gdb and gdbserver but I still get the same error about byte.

danboid, 2022-09-04:
The workaround to get this repo to build except uzem is to comment out line 15 ie

```
TOOLS += uzem
```

in the top `Makefile`

artcfox, 2022-09-21:
I fixed the build error by renaming the local typedef so it doesn't collide with the name defined by GCC 12 (and I renamed the other local typedefs for consistency). I noticed that I had the same exact build error on my Pinebook Pro running Manjaro ARM, and after I made the change I verified that uzem builds properly. Manjaro ARM doesn't seem to have a package that includes avr-gdb, so I used Debian 11 to verify that the gdbserver inside of uzem still works to debug into Uzebox game code. (Check the commit message for the commands I used to setup a remote debugging session and to set breakpoints in the game code.)

closed 2022-09-21 by artcfox