Uzebox/uzebox #105: Uzeamp: can't play wav files

danboid, 2023-02-26:
I have created a few wav files encoded using the ffmpeg command that Uze recommends on the Uzeamp wiki page https://uzebox.org/wiki/Uzeamp ie

ffmpeg -i input.mp3 -f wav -bitexact -acodec pcm_u8 -ac 1 -ar 15734 output.wav

On my real Uzebox, I have not been able to get Uzeamp to read anything from any of the SD cards I've tested with it, all were formatted with FAT16 on partitions smaller than 2 GB. I tried with 1 GB and 500 MB partitions but none of them could be seen by Uzeamp. The Uzebox bootloader can always read my (FAT16 or FAT32 formatted) SD card fine but Uzeamp cannot.

Under cuzebox I can get Uzeamp to at least list my wav files but I haven't been successful in getting anything to play. I push START on the wav file and there seems to be a brief pop sound  but the wav files do not start to play.

uze6666, 2023-02-27:
Uzeamp only support regular SD card under 2G. Anything over 2G like SDHC or SDXC functions differently and wont initialize. I can check to see if I could update the code to use FatFs. I prefer not to depend on the bootloader.

danboid, 2023-02-28:
As I said on the forums:

https://uzebox.org/forums/viewtopic.php?f=5&t=11236

The uzeamp.uze fil you posted there works for playing my wav files under cuzebox but non of the 5 or 6 SD cards I have tried formatting as FAT16 have worked for me under Uzeamp on my real Uzebox.

I'll change the subject of this bug to reflect the real issue.

artcfox, 2023-07-07:
> Uzeamp only support regular SD card under 2G. Anything over 2G like SDHC or SDXC functions differently and wont initialize. I can check to see if I could update the code to use FatFs. I prefer not to depend on the bootloader.

If you use bootlib.h, it will include FAT16 support, so even without the new bootloader, it will still function using FAT16 formatted SDSC cards. If you are using Bootloader5, then the same code will transparently support both FAT16 and FAT32 formatted SDSC and SDHC cards.

danboid, 2023-07-07:
I may have a go at fixing Uzeamp some time to do that then.

danboid, 2023-07-08:
Maybe not. The current Uzeamp code cannot play back wav files that the Uzeamp2.uze build on the wiki can play.

weber21w, 2023-07-10:
I had a look at this, but then found the changes required are non-trivial. It might need some fixes in mmc.s for the initialization to work, I don't know. Think I'll leave that to Alec's discretion personally. It would be nice to have this working though.

The kernel [mmc.h] [mmc_lib.c] and [mmc.s] could either go into the Uzeamp directory, or Uzeamp have it's functionality updated. The waters for SD solutions are pretty vast, PFF, FatFS, sdSimple, mmc_lib, bootlib...I don't think new developments are likely to use mmc_lib?

artcfox, 2023-07-10:
It's not just about the initialization, it's also a completely different filesystem that needs to be parsed differently.

I think the best option going forward would be to use bootlib.h, so you get a clean SDSC/FAT16 implementation as part of bootlib.h (built into the application), and as a bonus if you run that .uze on a Uzebox that has Bootloader5, it will automatically and transparently support both SDSC/FAT16 and SDHC/FAT32.

I had also taken a quick look at the code, and it looks like it would need a pretty big rewrite to be switched over to use bootlib.h.

danboid, 2023-08-08:
bootlib doesn't seem to do much yet, for example I don't see a function to list files in a directory?

petitfatfs apparently only supports a single file so thats probably no use if so.

https://ww1.microchip.com/downloads/en/Appnotes/Petit-Fat-File-System-00002799A.pdf

fatfs would seem like my best bet if I'm going to have a stab at this. I have found this documentation. Do you think most if not all the the SPI related stuff here applies to the fatfs we already have in the Uzebox repo?

http://elm-chan.org/fsw/ff/00index_e.html#:~:text=FatFs%20is%20a%20generic%20FAT%2FexFAT%20filesystem%20module%20for,8051%2C%20PIC%2C%20AVR%2C%20ARM%2C%20Z80%2C%20RX%20and%20etc

Because dealing with filesystems on this level is all new to me, I should try to port a program like this to the Uzebox first as a sanity check, see "Example SD card with SDIO"

http://stm32f4-discovery.net/2014/07/library-21-read-sd-card-fatfs-stm32f4xx-devices/

Maybe thats not a good example because its fatfs SDIO instead of SPI, we'd need to use SPI right? I'd also have to strip out all the stm32f4 specific bits of course.

danboid, 2023-08-13:
I'm wondering why fatfs was included in the Uzebox kernel dir when nothing uses it yet? Obviously it is useful for Uzeamp and who knows what else (a tracker, a pixel editor...).

Before I start trying to re-do Uzeamp to use fatfs, might it be worth updating the kernel fatfs to the latest version? I've noticed the Uzebox kernel fatfs source (ff.c) is R0.09b from 2013 whilst the latest version on the fatfs website is R0.15 from 2022.

http://elm-chan.org/fsw/ff/updates.html

danboid, 2024-09-04:
Thanks to @uze6666 for fixing this!

Uzeamp can now successfully read files from my 2 GB fat16 partition on a 32 GB MicroSD card but it is pretty unstable at playing any wavs that have been encoded for it and it has a few limitations regarding files too that need to be fixed but we can open specific bugs and requests for those.

closed 2024-09-04 by danboid