Uzebox/uzebox #113: Uzem doesn't work with Daemonbite SNES -> USB adapter under Linux

danboid, 2023-07-08:
I have a Daemonbite SNES adapter which allows me to use my SNES / Uzebox controllers under Linux but it doesn't work with Uzem (or cuzebox).

If load CRLTEST2.UZE into Uzem under Debian, I get:

B is registered by uzem as X
A works
Y is registered as B
X is registered as Y
SELECT is registered as LEFT SHOULDER BUTTON
START is registered as RIGHT SHOULDER BUTTON

Pushing the d-pad and the shoulder buttons doesn't register any input from CRLTEST2.UZE.

danboid, 2023-07-08:
@artcfox suggested I push 7 then START on the controller to configure it, after starting uzem. I have tried this and I only get as far as

```
Map an axial input (y/n)?
```

and thats as far as I get. If I type 'y' and move the joystick... nothing, If I type 'n' nothing happens so I can't complete the controller config process and my changed settings don't seem to take effect.

Note I was only able to configure all the buttons in uzem by using my holy arcade stick. When using the SNES pad, uzem reacted too quickly and it would assign one button to many, I think.

artcfox, 2023-07-08:
Cuzebox uses standard SDL2 mappings, which are configured externally for SDL applications. If there is not a mapping for the Daemonbite SNES adapter, there are 4 different controller mapping tools listed here: https://github.com/gabomdq/SDL_GameControllerDB maybe you can create a mapping for that adapter.

I was able to create a mapping for an old obscure game controller that I found which previously didn't work with anything.

artcfox, 2023-07-08:
I don't have a DaemonBite SNES adapter, but someone here was able to create a config for another DaemonBite product: https://github.com/MickGyver/DaemonBite-Arcade-Encoder so I think it's possible, and it might be that the same mapping posted here: https://github.com/libsdl-org/SDL/issues/7043 works for you?

danboid, 2023-07-09:
Wow what a mess!

I've not got any of those tools to fully work so I've been on a ticket opening spree:

https://gitlab.com/ryochan7/sdl2-gamepad-mapper/-/issues/4

https://gitlab.com/ryochan7/sdl2-gamepad-mapper/-/issues/5

I've tried gamepad-tool and that has a very simillar issue to sdl2-GM where two of my dpad directions don't register.  gamepad-tool isn't open source unfortunately.

How do we build the official SDL2 tool, gamepadmap? Another ticket!

https://github.com/libsdl-org/SDL/issues/7955

artcfox, 2023-07-09:
This won't solve the issue for Uzem, but it will for Cuzebox:

For SDL2, the utility is called controllermap, and it can be found in the test directory.

```
sudo apt install libsdl2-dev
git clone https://github.com/libsdl-org/SDL
cd SDL
git checkout SDL2
cd test
./configure
make
```
Plug in your controller
```
./controllermap
```
Follow the instructions printed to the console.

danboid, 2023-07-09:
Thanks Artcfox!

I found it funny that you knew how to build the controller config tool but the main SDL developer didn't! :D 

The good news is that the SDL tool (controllermap) actually works with all DPad directions unlike the other tools and I now have my SNES pad fully working under cuzebox by adding these lines to `gamecontrollerdb.txt`:

```
03000000412300003780000001010000,LLC Arduino Micro,platform:Linux,crc:9027,a:b0,b:b1,x:b2,y:b3,back:b6,start:b7,leftshoulder:b4,rightshoulder:b5,dpup:-a1,dpdown:+a1,dpleft:-a0,dpright:+a0,

03000000412300003780000001010000,LLC Arduino Micro,platform:Linux,crc:9027,a:b0,b:b1,x:b2,y:b3,back:b6,start:b7,leftshoulder:b4,rightshoulder:b5,dpup:-a1,dpdown:+a1,dpleft:-a0,dpright:+a0,
```

closed 2023-07-09 by danboid

artcfox, 2023-07-09:
> Thanks Artcfox!
>
> I found it funny that you knew how to build the controller config tool but the main SDL developer didn't! :D
>
> The good news is that the SDL tool (controllermap) actually works with [...]

Just a FYI, both of those lines are identical, so you only need to add one of them to the gamecontrollerdb.txt file.