Uzebox/uzebox #118: Suggestions for assembly guide

danboid, 2023-08-20:
The Uzebox has a great assembly guide PDF available here:

http://uzebox.org/files/UzeboxAssemblyGuideV1.7.pdf

I'm now most of my way through my second Uzebox build so I wanted to make a note a couple of things I'd like to see get added or changed in the guide.

This guide assumes that you have a Uzebox PCB with the AD725 chip already attached. That is unlikely to be the case.

I'm not sure exactly when might be optimal to attach the AD725 but it is best done before assembling the video circuitry on p14 because adding those caps makes soldering the AD725 more difficult, as someone has just realised.

It would be good if the guide got updated to give some tips regarding attaching the AD725 for those new to SMD soldering like myself. When I came to do this the first time, I didn't think about using blu tack to hold the chip in place so I  had he fantastic idea of supergluing it onto the PCB in a moment of poorer judgement because I didn't have a suitable clamp to hold it still as I soldered it. Blu tack is far from ideal but its something most people are likely to have in the house and its better than nothing if you don't have any better aids. Maybe you have a better and equally affordable technique?

Jimmy Dansbo uploaded this YT video to document how he attached the AD725 and SD card socket to his Uzebox so the assembly guide could link to Jimmy's video.

https://www.youtube.com/watch?v=S7GWMNdyopo

The guide also presumes that the user is using a pre-programmed ATMega. I think one of the steps on p13 needs to be updated to also add the SPI header and link to the avrdude Uzebox wiki page for instructions on flashing the MCU.

danboid, 2023-08-22:
Another thing I'd like to see added to the Uzebox docs that doesn't seem worth opening a separate ticket for is I'd like to see RS Components part numbers added to the BOM for European users. It's quite costly to get stuff shipped from Digikey to the UK.

danboid, 2023-08-24:
I hope you don't mind me posting random small issues with the Uzebox site here. I have another issue that doesn't seem quite worth opening a separate ticket for.

On https://uzebox.org/download.htm , the download link for the RGBUzebox schematics (note that schematic is spelled incorrectly too) is broken but you can still download the zip from:

https://kernelhacks.blogspot.com/2012/04/rgbuzebox.html

danboid, 2023-09-04:
It seems that the white RCA jack listed on the BOM is no longer available from Digi-key but this part seems to be identical:

KLPX-0848A-2-W

https://www.digikey.co.uk/en/products/detail/kycon-inc/KLPX-0848A-2-W/9990120

danboid, 2023-09-06:
Uze has posted a enclosure assembly guide on the forum which should be made available to download from http://uzebox.org/download.htm

https://docs.google.com/document/d/1DEXRXDNRst2V-QGHEevslhehpl1SDQHs3S_KmiEHp7o/edit

The guide says you need a hot glue gun but I'd say you'd quite easily get away without one, a tube of glue and some kind of applicator should suffice but thats a nitpick.

The enclosure guide recommends 6x 2mmx10mm self tapping screws but I used 3x12mm self tapping screws and they'd seem like a better choice to me.

I think it would be good to add some recommendations for those getting the shell 3D printed to the enclosure guide. I printed my first Uzebox top shell upside down. This has the advatage of sparing you having to print and remove any internal supports but it also prevents the ridges on the top printing correctly. I used my fingernail to eke out the ridges on the top. I'm not recommending that others also do this but it is an option if you care less about the looks and don't want the hassle of having to remove the supports.

You've probably got some much better 3D printing recommendations than me I expect.

danboid, 2023-09-14:
I would recommend that the assembly guide start with soldering the power jack first instead of the regulator. The reason for this is that the power jack is pretty much the only component that can be soldered on at the wrong angle. I spent 2 hours last night trying to unsolder the power jack from my latest build and I've failed to remove it so I've decided to start over with another board.

Because this was the third component I attached to the PCB, I've now lost three components instead of potentially just the one if that had been the first component I had been advised to attach.

danboid, 2023-11-21:
p8 has a "Other parts required" section and I think this should include a svideo or composite video cable and a SNES controller.

danboid, 2023-11-25:
It's turning out to be quite the epic issue this now isn't it?

At the end of the first post in this issue I said "The PDF also presumes that the user is using a pre-programmed ATMega. I think one of the steps on p13 needs to be updated to also add the SPI header and link to the avrdude Uzebox wiki page for instructions on flashing the MCU. At the stage where the user can test the audio there is enough of the Uzebox in place to use the SPI header to flash it."

Well it depends. If the user does have a pre-programmed MCU, then they are best off waiting until they have soldered the SD card socket before attaching the ISP header. They won't be able to solder the side of the SD socket nearest the ISP header if they install that first so I think this should be mentioned on page 13 ie it is recommended to solder the SD card socket before you solder the ISP header.

danboid, 2023-11-25:
p18 says:

"Add the RESET and POWER switches, R2(301Ω) and LED1. The longest pin on the LED is the anode and must be inserted in the hole with a little "+" next to it."

I'm pretty sure, having messed this up at least twice now, that the LED needs to be inserted the other way around ie the longest pin through the negative hole.

danboid, 2023-12-31:
The assembly guide could (should?) recommend the use of flux when soldering the SD card socket, the AD725 and the SNES connectors.

Doing so might be obvious to those experienced at soldering but this guide will also be read by those with no previous experience.

I have tried soldering the SD card socket and the SNES connectors without using flux and it did not go so well.

danboid, 2024-01-31:
Seeing as I've now almost finished my "How to build a Uzebox Omega" video, I'm pretty sure this will be the last suggestion I have to make for the assembly guide, my fine tooth comb having pretty much worn out.

The "Other parts required" section should list a AVR ISP mkii, one of the other programmers supported by avrdude or alternately 6x female to female Dupont cables if using GPIO.

weber21w, 2024-04-01:
Alec will have to look at this to fix the downloads page. Would adding assembly guides in the repo help? Would probably be easier to keep them up to date.

danboid, 2024-04-01:
Adding the assembly guides to the repo would make sense to me. Ideally they would be stored in a format easily editable with an open source, cross platform editor.

uze6666, 2024-08-13:
Lee's idea makes lots of sense, it will make it easier for anyone to update . And Markdown is easy to edit and open source. I've [imported the existing v1.7 assembly guide in a branch](https://github.com/Uzebox/uzebox/blob/AddUzeboxAssemblyGuide/schematics/Uzebox/V5.0/Uzebox-V5.0-assembly-guide.md#AVR-ISP-Programmers). Check it out and let me know what you think. It's obvious it's showing its age and will have to be redone at some point. All pics shows a very old rev of the PCB and the initial guide was geared toward kit buyers as Dan mentioned. I won't have time to do that in the short time but it's there if someone wants to tackle it.

danboid, 2024-08-14:
I'll check that branch out and add my suggestions soon.

danboid, 2024-08-28:
I'm still using Firefox as my main browser so I've installed this extension to get markdown previews rendered in FF outside of GitHub:

https://addons.mozilla.org/en-US/firefox/addon/markdown-viewer-webext/

danboid, 2024-08-29:
I have applied pretty much all of my suggestions above and a few more to this edit of the assembly guide:

[Uzebox-V5.0-assembly-guide.md](https://github.com/user-attachments/files/16790296/Uzebox-V5.0-assembly-guide.md)

Once @uze6666 has added that file and the accompanying images etc to the repo we can close this ticket.

danboid, 2024-08-29:
The only suggestion I've not added yet is this:

[quote]The "Other parts required" section should include a svideo or composite video cable and a SNES controller.[/quote]

Also solder flux to help soldering the AD725, SD card socket and SNES connectors. Due to the git workflow, I'd find it easier to add those after Alec has merged the guide into the repo.

I have added my recommendations to use flux to the guide already, its just not been added to the "Tools required" section yet.

uze6666, 2024-09-03:
> I have applied pretty much all of my suggestions above and a few more to this edit of the assembly guide: [...]

Could you please make a pull request on the branch for your changes, it will be easier for me to review the changes.

danboid, 2024-09-03:
I honestly don't know how to do that. Its probably time I learned to use git properly.

Alternately, you could compare the two using meld:

https://meldmerge.org/

uze6666, 2024-09-17:
Your suggestions have been merged into master.

danboid, 2024-09-17:
Great! I shall add a couple more pics / items to the tools and other parts section later and then we can close this.

danboid, 2024-09-17:
I have created https://github.com/Uzebox/uzebox/pull/143 to add in my final suggestions for the assembly guide.

danboid, 2024-09-23:
I updated my repo to fix a table formatting error.

uze6666, 2024-09-26:
I merged your latest changes, we can close this issue.

danboid, 2024-09-27:
Yes we can!

To think, essentially hundreds of hours are behind these few small changes but now this guide is fully ready for the newest of electronics newbs. They still have to be able to read English but that's about it now :D 

Thanks

closed 2024-09-27 by danboid