Uzebox/uzebox #119: Haiku support

danboid, 2023-09-02:
I prefer Haiku to Linux in many ways and I would like to be able to switch to it. It currently has bigger fish to fry in that respect with missing features such as HW virtualization and lacking proper support for most modern graphics hardware but one Haiku goal that we can achieve soon is making Haiku a supported Uzebox dev platform.

Haiku is already 99% of the way to be being a great Uzebox dev platform. It already has ports of avr-gcc, GNU tools, avrdude, vim, kate etc etc and both Uzem and cuzebox already build and run under Haiku so its just the Uzebox tools we have to add Haiku support for which I imagine should be pretty easy.

If I run `make` in the root dir of the Uzebox repo under Haikur1b4 it successfully builds Uzem but then the packrom makefile errors saying Haiku is an usupported platform. Haiku supports POSIX so it should be easy enough to get the tools building with a few tweaks to the Makefiles.

Before you try building the repo under Haiku, run:

```
pkgman update
<reboot>
pkgman install avr_gcc libsdl2_devel
```

To install the repo deps.

danboid, 2023-09-19:
I have also tried building the Uzebox repo under FreeBSD but Haiku gets further because Haiku at least manages to build Uzem before it errors. FreeBSD errors early in the Uzem build process but I don't recall what the exact error is.

It would be cool to see someone add (Free/Net/Open) BSD support to the Uzebox repo.

weber21w, 2024-04-01:
I don't know if there are others on the forum who use *BSD or Haiku, but since they are POSIX this might work?(I have no way to test):
//Add this wherever under Linux case in the Uzem makefile:

## NetBSD #############################
ifeq ($(OS),NetBSD)
OS := LINUX
PLATFORM := Unix
CC := g++
MKDIR := mkdir -p
RM := rm -rf
MTOOLS = 
endif

## FreeBSD #############################
ifeq ($(OS),FreeBSD)
OS := LINUX
PLATFORM := Unix
CC := g++
MKDIR := mkdir -p
RM := rm -rf
MTOOLS = 
endif

## OpenBSD #############################
ifeq ($(OS),OpenBSD)
OS := LINUX
PLATFORM := Unix
CC := g++
MKDIR := mkdir -p
RM := rm -rf
MTOOLS = 
endif

## Haiku #############################
ifeq ($(OS),Haiku)
OS := LINUX
PLATFORM := Unix
CC := g++
MKDIR := mkdir -p
RM := rm -rf
MTOOLS = 
endif

danboid, 2024-04-01:
Making your suggsted change to the packrom and gconvert makefiles and removing the `size:` build target from a few of the demos enables the Uzebox repo to build under Haiku if you install the avr_gcc, avr_libc and avr_binutils Haiku packages.

https://github.com/Uzebox/uzebox/pull/131

artcfox, 2024-04-01:
The size option comes from the vendor (Microchip) patched build tools, and
shows useful information for developers.

If you download the source for the toolchain from Microchip, you may be
able to compile the patched version for Haiku and preserve the size option.

Alternatively, instead of removing it, you could add a different "size"
target and conditionally enable the alternative for platforms that don't
use the official toolchains.

Or, the easiest would be to just put a dash in front of the command in the
Makefile, so it doesn't fail the build if it returns an error:

size:
    -avr-size -C

Cheers,
Matt (Artcfox)

danboid, 2024-06-21:
I've got the repo building under both Haiku and FreeBSD with a few small modifications to some of the Makefiles:

https://github.com/Uzebox/uzebox/pull/134

danboid, 2024-06-30:
I have updated https://uzebox.org/wiki/Getting_Started_on_the_Uzebox with links to new pages for build instructions for FreeBSD and Haiku

closed 2024-06-30 by danboid