Uzebox/uzebox #126: Uzenet issues

danboid, 2024-02-05:
I've already mentioned the following issues on the forum but I thought it made sense to log them here too.

The link to the Uzenet schematics and gerbers zip file on the downloads page is broken, I get a 404 error:

http://belogic.com/uzebox/schematics/Uzenet-module-v2.0.zip

I'm sure that there was an assembly guide for the Uzenet module? Maybe its in that zip file I can't download but otherwise that should be added to the Downloads page too as I cannot find it on the forum or in my inbox so I dunno where I got it.

The Uzenet config tool lists itself as UzeboxPC in the bootloader so I'd like to see that fixed.

The Uzenet config tool should print signal strength after connecting and there should be a short, memorable command to print the signal strength if it isn't already displayed.

danboid, 2024-02-10:
TBH I don't think the Uzenet module really needs a assembly guide, you can't really go wrong if you have bought it as a kit, provided you don't mess the soldering up.

It is a problem tha the gerber files aren't available for download because I've not been able to find them anywhere. Maybe they could be added to the main repo?

danboid, 2024-08-09:
The Uzenet module gerber files have been uploaded to the github repo.

uze6666, 2024-08-09:
And the links from the uzebox.org/downloads page now redirects to github so I'll close this issue.