Uzebox/uzebox #130: Update Downloads page

danboid, 2024-03-19:
The Uzebox Downloads page

https://uzebox.org/download.htm

needs to be updated.

The main thing that needs to be updated is the 3D printed enclosure section which currently only links to the old Uzebox case that was designed for the Type 1 SNES connectors.

This section should be updated to explain that if people have built a Uzebox or a Uzebox Omega using Type 2 connectors they should use the .3mf files in the Uzebox repo rather than printing the files hosted on Thingiverse or Youmagine.

You could also link to the wiki page I created for people to share tips on printing and modifications to the case:

https://uzebox.org/wiki/3D_printed_case

The other issues with the DL page that I recall are that the link to the Uzenet module gerbers zip file is broken:

http://belogic.com/uzebox/schematics/Uzenet-module-v2.0.zip

Also I recall there was an assembly guide for building the Uzenet module but that isn't linked to on the DL page nor can I find it on the forum or in my inbox. Am I imagining that PDF?

weber21w, 2024-04-01:
I don't recall if there was an assembly guide for the Uzenet module.

I know Alec put some new items in the repo recently under cad/enclosure, not sure if that's what you want. If you have something to add to the repo for this can you do a pull request?

danboid, 2024-04-01:
The source code for the Uzebox site (like the Downloads page) isn't in the Uzebox repo so I cannot do a PR for these changes.

danboid, 2024-04-24:
WRT the Uzebox case, I think the best case to print depends on which type of SNES connectors you are using.

If you are using type-1 connectors, people should print the files hosted on Thingiverse but those using type 2 connectors should print the files in the repo. Is that correct Alec?

danboid, 2024-05-23:
Thorsten got hit by this recently too. I had to point him towards this bug report and tell him which files he should've printed today.

danboid, 2024-08-08:
@uze6666 has fixed part of this issue by uploading the gerbers for the Uzenet module to the github repo.

danboid, 2024-08-09:
Yep, looks good to me now.

Thanks Alec!

closed 2024-08-09 by danboid

danboid, 2024-08-09:
Oops!

Sorry. I didn't really check the page so I opened a new ticket for the new errors:

https://github.com/Uzebox/uzebox/issues/137