Uzebox/uzebox #132: Fix sound engine

danboid, 2024-04-04:
IKD's Commando theme remake unveiled a bug in the Uzebox sound engine that caused it not to play properly on real Uzeboxes.

Uze said he wanted to do a bit more testing before adding it to the repo so I merged the fix myself to upload an IKD v1.1 build with fixed music playback but I think this fix is important enough that it should get upstreamed to stop similar issues re-ocurring with future games should anyone else feel the urge to push the boat out with a fancy Uzebox chiptune. We don't want a buggy sound engine as standard.

uze6666, 2026-09-29:
Finally resolved by pull request #193. In pull request there's a readme in the zip that explains the issue and fix. It *may* have effects on games made after 2018 that designed sound patches that relied on the bug, but it would be surprising since it was resulting in a lock-up of the LFSR and mainly muted the channel (or only made clicks).
@danboid : Ionfirm if fixes IKD and close. You wont need my old C fix anymore (see the zip).

danboid, 2026-09-30:
Thanks Alec.

I'll probably do a PR to add IKD to the Uzebox repo after I've tested this.

Might as well close this now.

closed 2026-09-30 by danboid