Uzebox/uzebox #133: Cannot build Uzebox repo under Windows

danboid, 2024-04-20:
I tried building the Uzebox repo under Windows for the first time today but I didn't get very far.

I followed the instructions here:

https://uzebox.org/wiki/Windows_Software_Installation

The first issue is that the instructions don't cover installing SDL2 which is required to build Uzem (and cuzebox).

[image]

I tried downloading this:

https://github.com/libsdl-org/SDL/releases/download/release-2.30.0/SDL2-2.30.0-win32-x64.zip

I extracted SDL2.dll from that zip and copied it into C:\SDL2 but I still get thesame error about SDL library not found.

I tried commenting out Uzem from the Makefile to try building the repo without it but I only got as far as this:

[image]

danboid, 2024-04-20:
We might want to update the gcc links too? I see avr-gcc has v13.2.0 out

https://github.com/ZakKemble/avr-gcc-build/releases/tag/v13.2.0-1

gcc has a v14 release:

https://github.com/brechtsanders/winlibs_mingw/releases

I've not tested either.

artcfox, 2024-04-20:
I believe that we should be recommending the official avr-gcc toolchains
that have all of the vendor patches:
https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers

weber21w, 2024-04-21:
Not sure if it's right, but I posted a workaround on SDL2/Windows for Uzem:
https://uzebox.org/forums/viewtopic.php?p=37423

Can't build the entire repo on my end, need to investigate issues with tools/bin2hex

danboid, 2024-04-22:
SDL2.dll needs to be copied into c:\SDL2\bin and I had to copy the include files into c:\SDL2\include\SDL2\SDL2

Now its getting ld to work...

danboid, 2024-04-22:
I noticed that uzem wants to build the debug build so I removed debug from TARGETS in line 14 of the uzem Makefile so that it only builds the release version of uzem. This screenshot shows the ld error I'm getting with everything right now:

[image]

danboid, 2024-04-23:
I have got Uzem and all of the tools building under Windows now but I've done it in a rather hacky way hence no PR.

The main Makefile needs to be edit this, around line 103:

```
#DEST_FLAG = DEST_DIR=$(shell pwd)/$(BIN_DIR)/
#DEMO_FLAG = UZEBIN_DIR=$(shell pwd)/$(BIN_DIR)/
DEST_DIR = C:\tools\bin
UZEBIN_DIR= C:\tools\bin
```

Replace `C:\tools\bin` with a sensible path. The current Makefile is trying to use some nonsensical path to install stuff like `/c/blah/blah` so of course ld.exe and friends get upset.

danboid, 2024-05-24:
The best workaround for Windows users until this gets fixed is to install and build the repo under WSL.

https://learn.microsoft.com/en-us/windows/wsl/install

danboid, 2024-05-24:
I have edited

https://uzebox.org/wiki/Windows_Software_Installation

To include a warning that the repo currently doesn't build under Windows and to use WSL instead.

danboid, 2024-07-03:
I have been able to build all of the Uzebox repo (tools and demos) for Windows using msys2 but I'm doing something wrong with uzem. It builds but when I try to run uzem Windows complains about it not finding the SDL2 dll.

It seems we're probably better off rewriting the current Windows getting started guide to use msys2 instead of msys, like the current guide uses:

https://uzebox.org/wiki/Windows_Software_Installation

Its easier to get the dev tools installed under msys2 thanks to pacman.

msys2 has several environments to choose from. It looks like we need to pick between ucrt64 and mingw64. The different msys2 environments are explained here:

https://www.msys2.org/docs/environments/

They say "If you are unsure, go with UCRT64."

artcfox, 2024-07-03:
The instructions for building are wrong, they need the -devel version of
the SDL2 libs

And you can just put them in the same dir as the executable and it will
find them.

I have built the Windows version of cuzebox before (details in some thread
on the forum) and I think the Mac version, using cross compilers on Linux.
Then I tested it using wine, lol.

danboid, 2024-07-03:
This is important to understand msys2 and how it handles paths:

https://www.msys2.org/docs/filesystem-paths/

danboid, 2024-07-08:
Today I successfully built and tested cuzebox under Windows using msys2 so I've potentially fixed the Windows build if we take option one below.

Jubatian has said he's done with maintaining cuzebox and I've not heard any objections to merging it into the Uzebox repo, it would be silly not to.

I don't think it makes any sense to build both uzem and cuzebox when building the repo so by my thinking we have two options:

1 - We add cuzebox to the repo and modify the repo Makefile so that NO emulators are built by default. This is my preferred option because the emulators don't get updated very often and most of the time all I really want to build are the tools. It makes sense to carry on building all the demos too and thats fine because that works under msys2 as is. If we take this option, I can rewrite the Windows repo install instructions and we can fix the Windows build by simply removing Uzem from the main Makefile. The user would then have to manually build either cuzebox, uzem or both.

2 - We add cuzebox and modify the main makefile to build cuzebox by default. This requires modifying both the main Makefile and the Makefile for cuzebox so that it can auto-detect when building under Windows. It shouldn't be hard to do but I've not done it yet.

I don't see the point in spending much time fixing Uzem's Makefile when its mostly been replaced by cuzebox and we are going to add cuzebox to the repo anyway and I expect at that point it will replace uzem as the default emulator, if we choose to still build an emulator by default.

Either way, I think its best I rewrite the Windows install guide to use msys2 as it seems like the easiest way to build the Uzebox repo under Windows for Windows.

artcfox, 2024-07-08:
If anything, we should add cuzebox to the main repo and have it built along
with uzem.

We want all the tools to be built by default.

Uzem can do useful things that cuzebox cannot, namely remote debugging with
its built in gdbserver, and whisper port printing, input capture/playback.

Some of those features I have added to my fork of cuzebox under the
debug-enhancements branch, but not all (like the gdbserver)

danboid, 2024-07-08:
I haven't been able to build uzem under Windows with gdbserver enabled, only with
`
NOGDB=1 make release`

danboid, 2024-07-08:
I have tried installing `gdb`under msys2 but that doesn't fix building the debug version of uzem.

```
Dan@DESKTOP-7B3CGMT UCRT64 ~/uzebox
$ make
make -C tools/uzem  DEST_DIR=/home/Dan/uzebox/bin/
make debug
=================================
Building Debug...
Platform: Windows-Windows_NT
=================================
g++ -c uzem.cpp -o Debug/uzem.o -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzem.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c -S uzem.cpp -o Debug/uzem.o.asm -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzem.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c avr8.cpp -o Debug/avr8.o -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/avr8.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c -S avr8.cpp -o Debug/avr8.o.asm -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/avr8.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c uzerom.cpp -o Debug/uzerom.o -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzerom.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c -S uzerom.cpp -o Debug/uzerom.o.asm -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzerom.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
g++ -c gdbserver.cpp -o Debug/gdbserver.o -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/gdbserver.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
In file included from C:/msys64/ucrt64/include/wtypes.h:8,
                 from C:/msys64/ucrt64/include/winscard.h:10,
                 from C:/msys64/ucrt64/include/windows.h:97,
                 from C:/msys64/ucrt64/include/winsock2.h:23,
                 from gdbserver.h:31,
                 from gdbserver.cpp:36:
C:/msys64/ucrt64/include/rpcndr.h:64:11: error: reference to 'byte' is ambiguous
   64 |   typedef byte cs_byte;
      |           ^~~~
In file included from C:/msys64/ucrt64/include/c++/13.2.0/bits/memory_resource.h:38,
                 from C:/msys64/ucrt64/include/c++/13.2.0/string:58,
                 from C:/msys64/ucrt64/include/c++/13.2.0/bits/locale_classes.h:40,
                 from C:/msys64/ucrt64/include/c++/13.2.0/bits/ios_base.h:41,
                 from C:/msys64/ucrt64/include/c++/13.2.0/ios:44,
                 from C:/msys64/ucrt64/include/c++/13.2.0/ostream:40,
                 from C:/msys64/ucrt64/include/c++/13.2.0/iostream:41,
                 from gdbserver.cpp:26:
C:/msys64/ucrt64/include/c++/13.2.0/cstddef:69:14: note: candidates are: 'enum class std::byte'
   69 |   enum class byte : unsigned char {};
      |              ^~~~
C:/msys64/ucrt64/include/rpcndr.h:63:25: note:                 'typedef unsigned char byte'
   63 |   typedef unsigned char byte;
      |                         ^~~~
C:/msys64/ucrt64/include/rpcndr.h:397:170: error: reference to 'byte' is ambiguous
  397 |   typedef void (__RPC_API *CS_TYPE_TO_NETCS_ROUTINE)(RPC_BINDING_HANDLE hBinding,unsigned __LONG32 ulNetworkCodeSet,void *pLocalData,unsigned __LONG32 ulLocalDataLength,byte *pNetworkData,unsigned __LON
G32 *pulNetworkDataLength,error_status_t *pStatus);
      |                                                                                                                                                                          ^~~~
C:/msys64/ucrt64/include/c++/13.2.0/cstddef:69:14: note: candidates are: 'enum class std::byte'
   69 |   enum class byte : unsigned char {};
      |              ^~~~
C:/msys64/ucrt64/include/rpcndr.h:63:25: note:                 'typedef unsigned char byte'
   63 |   typedef unsigned char byte;
      |                         ^~~~
C:/msys64/ucrt64/include/rpcndr.h:397:170: error: 'byte' has not been declared
  397 |   typedef void (__RPC_API *CS_TYPE_TO_NETCS_ROUTINE)(RPC_BINDING_HANDLE hBinding,unsigned __LONG32 ulNetworkCodeSet,void *pLocalData,unsigned __LONG32 ulLocalDataLength,byte *pNetworkData,unsigned __LON
G32 *pulNetworkDataLength,error_status_t *pStatus);
      |                                                                                                                                                                          ^~~~
C:/msys64/ucrt64/include/rpcndr.h:398:119: error: reference to 'byte' is ambiguous
  398 |   typedef void (__RPC_API *CS_TYPE_FROM_NETCS_ROUTINE)(RPC_BINDING_HANDLE hBinding,unsigned __LONG32 ulNetworkCodeSet,byte *pNetworkData,unsigned __LONG32 ulNetworkDataLength,unsigned __LONG32 ulLocalBu
fferSize,void *pLocalData,unsigned __LONG32 *pulLocalDataLength,error_status_t *pStatus);
      |                                                                                                                       ^~~~
C:/msys64/ucrt64/include/c++/13.2.0/cstddef:69:14: note: candidates are: 'enum class std::byte'
   69 |   enum class byte : unsigned char {};
      |              ^~~~
C:/msys64/ucrt64/include/rpcndr.h:63:25: note:                 'typedef unsigned char byte'
   63 |   typedef unsigned char byte;
[... cut ...]
  883 |             bval  = hex2nib(*pkt++) << 4;
      |             ^~~~
gdbserver.cpp:892:13: error: 'wval' was not declared in this scope
  892 |             wval  = hex2nib(*pkt++) << 4; /* low byte first */
      |             ^~~~
gdbserver.cpp:904:13: error: 'bval' was not declared in this scope
  904 |             bval  = hex2nib(*pkt++) << 4;
      |             ^~~~
make[2]: *** [Makefile:300: Debug/gdbserver.o] Error 1
make[1]: *** [Makefile:281: all] Error 2
make: *** [Makefile:136: tools/uzem] Error 2
```

danboid, 2024-07-08:
Unless we can fix Windows gdb support, wouldn't it make sense to build Uzem without debug support for Windows only rather than have the whole build fail because Uzem fails.

Alternately we could build Uzem last rather than first so that Windows users get all the other tools built before Uzem kills the repo build process.

artcfox, 2024-07-08:
That error might be the result of

using namespace std;

See: https://github.com/msys2/MINGW-packages/issues/7035

danboid, 2024-07-08:
I commented out

`using namespace std`

From avr8.h, avr8.cpp, gdbserver.h and gdbserver.cpp and now I get these errors when tryig to build uzem under msys2;

```
Dan@DESKTOP-7B3CGMT UCRT64 ~/uzebox
$ make
make -C tools/uzem  DEST_DIR=/home/Dan/uzebox/bin/
make debug
=================================
Building Debug...
Platform: Windows-Windows_NT
=================================
g++ -c uzem.cpp -o Debug/uzem.o -IC:/SDL2/include/SDL2 -LC:/SDL2/lib -DWindows_NT -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzem.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
In file included from avr8.h:33,
                 from uzem.cpp:27:
gdbserver.h:61:33: error: expected template-name before '<' token
   61 | class Breakpoints: public vector<dword_t> {
      |                                 ^
gdbserver.h:61:33: error: expected '{' before '<' token
gdbserver.h:61:33: error: expected unqualified-id before '<' token
gdbserver.h:153:21: error: field 'BP' has incomplete type 'Breakpoints'
  153 |         Breakpoints BP;
      |                     ^~
gdbserver.h:61:7: note: forward declaration of 'class Breakpoints'
   61 | class Breakpoints: public vector<dword_t> {
      |       ^~~~~~~~~~~
avr8.h:504:9: error: 'queue' does not name a type
  504 |         queue <u8> uzeKbScanCodeQueue;
      |         ^~~~~
make[2]: *** [Makefile:300: Debug/uzem.o] Error 1
make[1]: *** [Makefile:281: all] Error 2
make: *** [Makefile:136: tools/uzem] Error 2
```

danboid, 2024-07-09:
I get the same errors if I comment out `using namespace std;` under Linux

```
$ make
make -C tools/uzem  DEST_DIR=/home/dan/uzebox/bin/
make debug
=================================
Building Debug...
Platform: Unix-LINUX
=================================
g++ -c uzem.cpp -o Debug/uzem.o -I/usr/include/SDL2 -D_REENTRANT -DLINUX -D_GNU_SOURCE=1 -DGUI=1 -DJOY_ANALOG_DEADZONE=8192 -g -MD -MP -MF Debug/uzem.d -DUSE_SPI_DEBUG=1 -DUSE_EEPROM_DEBUG=1 -DUSE_GDBSERVER_DEBUG=1
In file included from avr8.h:33,
                 from uzem.cpp:27:
gdbserver.h:61:33: error: expected template-name before ‘<’ token
   61 | class Breakpoints: public vector<dword_t> {
      |                                 ^
gdbserver.h:61:33: error: expected ‘{’ before ‘<’ token
gdbserver.h:61:33: error: expected unqualified-id before ‘<’ token
gdbserver.h:153:21: error: field ‘BP’ has incomplete type ‘Breakpoints’
  153 |         Breakpoints BP;
      |                     ^~
gdbserver.h:61:7: note: forward declaration of ‘class Breakpoints’
   61 | class Breakpoints: public vector<dword_t> {
      |       ^~~~~~~~~~~
avr8.h:504:9: error: ‘queue’ does not name a type
  504 |         queue <u8> uzeKbScanCodeQueue;
      |         ^~~~~
make[2]: *** [Makefile:300: Debug/uzem.o] Error 1
make[1]: *** [Makefile:281: all] Error 2
make: *** [Makefile:136: tools/uzem] Error 2
```

danboid, 2024-07-09:
I have emailed Filipe to ask if he could have a look at this. He's Principle Software Engineer at ARM now.

artcfox, 2024-07-09:
Did you follow the link I sent and read the suggestion? You can't just
remove that "using" line without adding the std:: namespace to every spot
in the code that needs it.

I don't have access to a Windows computer, so I can't test if that works.

danboid, 2024-07-09:
Oh yes! Now I remember how C++ namespaces work. I don't actually know C++ but I do remember reading about namespaces.

As I said in a previous post, I think (hope) its only avr8.h, avr8.cpp, gdbserver.h and gdbserver.cpp that need refactoring.

I have found a tool that may be able to automate this for us under emacs called rtags:

https://github.com/Andersbakken/rtags

I have successfully built it under Debian but I've not tried it yet. I need to create JSON Compilation Database for Uzem first but that looks easy to do with bear:

https://github.com/rizsotto/Bear

As for Windows, you can easily install Windows 10 or 11 in a VirtualBox (or virt-manager/qemu) VM:

https://www.microsoft.com/en-gb/software-download/windows10ISO

You do not need a license to install Windows 10, just skip the installer screen that asks for a license key. The only disadvantage to running unlicensed Windows is you cannot customise the desktop and there will be an unlicensed notification on your desktop.

danboid, 2024-07-09:
```
dan@rigodon:~/src/rtags/bin$ [ 20%] 1/5 22:50:07 ~/uzebox/tools/uzem/uzerom.cpp in 455ms. (4 builds) (5398 syms, 4436 symNames, 125 includes, 73 of 73 files, symbols: 23626 of 23626, 30636 cursors, 3397338 bytes written, 72 queried 17ms) (115/175/35ms). (priority 0)
[ 40%] 2/5 22:50:08 ~/uzebox/tools/uzem/SDEmulator.cpp in 922ms. (4 builds) (6125 syms, 8179 symNames, 86 includes, 43 of 114 files, symbols: 54296 of 65470, 74228 cursors, 4091191 bytes written, 113 queried 16ms) (484/300/27ms). (priority 0)
[ 60%] 3/5 22:50:11 ~/uzebox/tools/uzem/uzem.cpp in 4514ms. (4 builds) (885 syms, 317 symNames, 13 includes, 6 of 319 files, symbols: 4258 of 105538, 110332 cursors, 326390 bytes written, 318 queried 11ms) (4308/176/5ms). (priority 0)
[ 80%] 4/5 22:50:12 ~/uzebox/tools/uzem/gdbserver.cpp in 5423ms. (4 builds) (3507 syms, 2287 symNames, 60 includes, 27 of 416 files, symbols: 17694 of 133168, 144812 cursors, 1738988 bytes written, 415 queried 24ms) (5130/231/15ms). (priority 0)
[100%] 5/5 22:50:19 ~/uzebox/tools/uzem/avr8.cpp in 12313ms. (4 builds) (158892 syms, 106685 symNames, 779 includes, 295 of 390 files, symbols: 765598 of 790434, 1114001 cursors, 88301625 bytes written, 389 queried 151ms) (3716/6515/389ms). (priority 0)
Jobs took 12.42s, (avg 2.48s). We're using 17mb of memory. 
Finish output from /home/dan/uzebox/tools/uzem/avr8.cpp
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:259:40: Replace 8 characters with "size(s)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:259:50: Replace 7 characters with "avail(s)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:10: Replace 16 characters with "cycleCounter(-1)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:28: Replace 12 characters with "elapsedCyclesSleep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:42: Replace 10 characters with "prevPortB(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:54: Replace 24 characters with "prevWDR(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:79: Replace 16 characters with "watchdogTimer(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:3: Replace 10 characters with "TCNT1(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:15: Replace 12 characters with "timer1_next(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:29: Replace 21 characters with "timer1_base(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:51: Replace 16 characters with "itd_TIFR1(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:317:3: Replace 18 characters with "dly_out(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:319:3: Replace 14 characters with "enableGdb(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:319:19: Replace 14 characters with "eepromFile("eeprom.bin")"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:319:35: Replace 8 characters with "hsyncHelp(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:322:3: Replace 16 characters with "recordMovie(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:328:3: Replace 17 characters with "inset(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:328:21: Replace 8 characters with "fullscreen(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:334:3: Replace 15 characters with "pad_mode(SNES_PAD)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:334:19: Replace 18 characters with "joystickFile(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:3: Replace 13 characters with "gdb(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:18: Replace 17 characters with "gdbBreakpointFound(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:37: Replace 25 characters with "gdbInvalidOpcode(false)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:63: Replace 23 characters with "gdbPort(1284)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:87: Replace 13 characters with "state(CPU_STOPPED)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:339:3: Replace 18 characters with "singleStep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:339:22: Replace 6 characters with "nextSingleStep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:349:15: Replace 11 characters with "spiTransfer(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:349:28: Replace 14 characters with "spiClock(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:259:40: Replace 8 characters with "size(s)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:259:50: Replace 7 characters with "avail(s)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:10: Replace 16 characters with "cycleCounter(-1)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:28: Replace 12 characters with "elapsedCyclesSleep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:42: Replace 10 characters with "prevPortB(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:54: Replace 24 characters with "prevWDR(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:314:79: Replace 16 characters with "watchdogTimer(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:3: Replace 10 characters with "TCNT1(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:15: Replace 12 characters with "timer1_next(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:29: Replace 21 characters with "timer1_base(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:315:51: Replace 16 characters with "itd_TIFR1(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:317:3: Replace 18 characters with "dly_out(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:319:3: Replace 14 characters with "enableGdb(false)"
[... cut ...]
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:63: Replace 23 characters with "gdbPort(1284)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:338:87: Replace 13 characters with "state(CPU_STOPPED)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:339:3: Replace 18 characters with "singleStep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:339:22: Replace 6 characters with "nextSingleStep(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:349:15: Replace 11 characters with "spiTransfer(0)"
Fixit for /home/dan/uzebox/tools/uzem/avr8.h:349:28: Replace 14 characters with "spiClock(0)"
```

danboid, 2024-07-09:
```
dan@rigodon:~/src/rtags/bin$ ./rc --help
rc options...
Options:
  --verbose|-v                           Be more verbose.
  --version                              Print current version.
  --verify-version|-t [arg]              Verify that the correct protocol version is used.
  --silent|-Q                            Be silent.
  --help|-h                              Display this help.
  --config [arg]                         Use this file (instead of ~/.rcrc).
  --no-rc                                Don't load any rc files.

Rdm:
  --quit-rdm|-q                          Tell server to shut down with optional exit code as argument.
  --connect-timeout [arg]                Timeout for connecting to rdm in ms (default 1000).

Project management:
  --clear|-C                             Clear projects.
  --project|-w [optional]                With arg, select project matching that if unique, otherwise list all projects.
  --delete-project|-W [arg]              Delete all projects matching regex.
  --job-count|-j [optional]              Set or query current job count. (Prefix with l to set low-priority-job-count).

Indexing commands:
  --compile|-c [optional]                Pass compilation arguments to rdm.
  --guess-flags                          Guess compile flags (used with -c).
  --load-compile-commands|-J [optional]  Load compile_commands.json from directory
  --suspend|-X [optional]                Dump suspended files (don't track changes in these files) with no arg. Otherwise toggle suspension for arg.

Query commands:
  --follow-location|-f [arg]             Follow this location.
  --references-name|-R [arg]             Find references matching arg.
  --references|-r [arg]                  Find references matching this location.
  --list-symbols|-S [optional]           List symbol names matching arg.
  --find-symbols|-F [optional]           Find symbols matching arg.
  --symbol-info|-U [arg]                 Get cursor info for this location.
  --status|-s [optional]                 Dump status of rdm. Arg can be symbols or symbolNames.
  --diagnose [arg]                       Resend diagnostics for file.
  --diagnose-all                         Resend diagnostics for all files.
  --last-indexed                         Get timestamp of the last time indexing completed for the current project.
  --is-indexed|-T [arg]                  Check if rtags knows about, and is ready to return information about, this source file.
  --is-indexing [optional]               Check if rtags is currently indexing files in any project or in project matching pattern.
  --has-filemanager [optional]           Check if rtags has info about files in this directory.
  --preprocess|-E [arg]                  Preprocess file.
  --asm [arg]                            Assemble file.
  --reindex|-V [optional]                Reindex all files or all files matching pattern.
  --check-reindex|-x [optional]          Check if reindexing is necessary for all files matching pattern.
  --path|-P [optional]                   Print files matching pattern.
  --current-project                      Print path for current project.
  --dump-file|-d [arg]                   Dump source file.
  --check-includes [arg]                 Check includes for source file.
  --dump-file-maps [arg]                 Dump file maps for file.
  --generate-test [arg]                  Generate a test for a given source file.
  --rdm-log|-g                           Receive logs from rdm.
  --fixits [arg]                         Get fixits for file.
  --find-project-root [arg]              Use to check behavior of find-project-root.
  --find-project-build-root [arg]        Use to check behavior of find-project-root for builds.
  --include-file [arg]                   Use to generate include statement for symbol.
  --sources [optional]                   Dump sources for source file.
  --dependencies [arg]                   Dump dependencies for source file [(includes, included-by, depends-on, depended-on, tree-depends-on, raw)].
  --all-dependencies                     Dump dependencies for all source files [(includes, included-by, depends-on, depended-on, tree-depends-on, raw)].
  --reload-file-manager|-B               Reload file manager.
  --code-complete-at|-l [arg]            Code complete at location: arg is file:line:col.
  --send-diagnostics [arg]               Only for debugging. Send data to all -G connections.
  --dump-completions                     Dump cached completions.
  --dump-compile-commands                Dump compilation database for project.
  --set-buffers [optional]               Set active buffers (list of filenames for active buffers in editor).
  --list-buffers                         List active buffers.
  --add-buffers [arg]                    Add additional buffers.
  --remove-buffers [arg]                 Remove buffers.
  --list-cursor-kinds                    List spelling for known cursor kinds.
[... 48 lines cut ...]
  --compilation-flags-only               For --source, only print compilation flags.
  --compilation-flags-split-line         For --source, print one compilation flag per line.
  --compilation-flags-pwd                For --source, print pwd for compile command on the first line.
  --dump-include-headers                 For --dump-file, also dump dependencies.
  --silent-query                         Don't log this request in rdm.
[... cut ...]
  --code-complete-prefix [arg]           Filter out code completion results that don't start with this prefix.
  --code-completion-enabled|-b           Inform rdm that we're code-completing. Use with --diagnose
  --no-spell-checking                    Don't produce spell check info in diagnostics.
  --tokens-include-symbols               Include symbols for tokens.
  --no-realpath                          Don't resolve paths using realpath(3).
  --include-path [arg]                   Dump include path for symbol.
  --max-depth [arg]                      Max search depth. Used for --include-path.
```

filiperinaldi, 2024-07-11:
Hi @danboid , It was 15 years ago when I worked on this and I don't remember the details. I don't think I ever tried it on Windows. It seems like you already realised the issue is with the std namespace. More specifically, based on the compilation errors, it seems a Windows specific header has also a "byte" definition which is clashing with the std definition being included by gdbserver.h. From what I can see the gdbserver.h/.cpp files only use std for the std::vector. Have you tried removing std from the gdbserver.h only and explicitly using std::vector?
Something like:

```
--- a/tools/uzem/gdbserver.h
+++ b/tools/uzem/gdbserver.h
@@ -50,15 +50,13 @@
 #include <vector>

-using namespace std;
-
 struct avr8;

 typedef uint8_t byte_t;
 typedef uint16_t word_t;
 typedef uint32_t dword_t;

-class Breakpoints: public vector<dword_t> {
+class Breakpoints: public std::vector<dword_t> {
 };
```

danboid, 2024-07-11:
I did as @filiperinaldi advised as well as adding

```
#include <cstdint>
```

to `gdbserver.h` and I also had to remove `using namespace std;` from `gdbserver.cpp` and now Uzem and hence the full Uzebox repo build without error under msys2 in my fork so I've opened a PR:

https://github.com/Uzebox/uzebox/pull/136

Thanks @filiperinaldi and @artcfox !

danboid, 2024-07-12:
I have updated the Windows repo install instructions wiki page with MSYS2 instructions:

https://uzebox.org/wiki/Windows_Software_Installation#Building_the_Uzebox_repository_using_MSYS2

It is using my repo currently but we can change that as soon as the main repo is fixed.

danboid, 2024-07-14:
I forgot to mention that I have tested uzem for Windows after building it using my repo and the instructions here:

https://uzebox.org/wiki/Windows_Software_Installation#Building_the_Uzebox_repository_using_MSYS2

I have just updated those instructions now to note that the user has to install SDL2.dll if they want to run uzem. You can build Uzem fine under msys2 with only the msys2 SDL2 dev package installed but without having SDL2.dll installed under c:\windows\system32.

uze6666, 2024-08-09:
I spawned a fresh new Windows 11 VM and followed the instructions. Pretty amazing that everything that was so painful to gather is so easily integrated now, cleary the way to go on Windows! Unfortunately the build fails immediately with:
[image]
though I can see SDL2 package is installed. I modified the Makefile to first check if running in an MSYS2 environment under Windows:
```
## Windows ###########################
#
# IMPORTANT: To build under MinGW, you will need file "SDL_platform.h" from SDL2 2.0.4 (currently a snapshot)
# because of a bug in SDL_platform.h in 2.0.3 (currently stable release)
# C:\MinGW\include\SDL2/SDL_platform.h:121:26: fatal error: winapifamily.h: No such file or directory
# Be sure to use SDL /i686-w64-mingw32 folder for 32 bit target
#
ifneq (,$(findstring MINGW,$(UNAME)))
PLATFORM := Windows
OS_EXTENSION := .exe

# First check if we are running in a MSYS2 / ucrt64 environment
SDL_BASE_DIR := /ucrt64
SDL_DIR_ENTRIES = $(wildcard $(SDL_BASE_DIR)/include/*)
ifeq ($(findstring SDL2,$(SDL_DIR_ENTRIES)),SDL2)
    SDL_FLAGS    = -I$(SDL_BASE_DIR)/include/SDL2 -L$(SDL_BASE_DIR)/lib
    SDL_DLL_FROM = $(SDL_BASE_DIR)/bin/SDL2.dll
endif

# Didn't find? Then check in C:/SDL2...
ifeq ($(SDL_FLAGS),)  
    SDL_BASE_DIR := C:/SDL2
    SDL_DIR_ENTRIES = $(wildcard C:/SDL2/*)
    ifeq ($(findstring include,$(SDL_DIR_ENTRIES)),include)
        ifeq ($(findstring bin,$(SDL_DIR_ENTRIES)),bin)
            SDL_FLAGS = -I$(SDL_BASE_DIR)/include/SDL2 -L$(SDL_BASE_DIR)/lib
            SDL_DLL_FROM = $(SDL_BASE_DIR)/bin/SDL2.dll
        endif
    endif
endif

.....
```
After that all tools build perfectly without adding or moving any DLL. Through games compilation failed at Mode13ExtendedDemo. I will resume later.

uze6666, 2024-08-09:
Ok was missing an explicit cast.

In /kernel/videoMode13/videoMode13.c: line 109
Replace 
`u8* src = (sprites[sprNo].tileIndex*(TILE_HEIGHT*TILE_WIDTH/2))`
With
`u8* src =(u8*)(sprites[sprNo].tileIndex*(TILE_HEIGHT*TILE_WIDTH/2))`

After you add this and the fix to the main Makefile in your fork, let me know and I'll pull the changes and re-test everything.

danboid, 2024-08-10:
I have added those changes to my fork now but unfortunately I also synced my fork to the main repo which has pulled in Lee's recent changes which means building the repo only gets as far as the unittest demo.

danboid, 2024-08-11:
Lee has fixed the repo build issue so I synced my repo again and it builds under both Windows 10 and Debian 12 again now.

uze6666, 2024-08-13:
Builds perfectly on my windows 11 now. You can submit pull request. Anyone who gets a chance can merge it.

danboid, 2024-08-13:
https://github.com/Uzebox/uzebox/pull/136

uze6666, 2024-08-13:
Done!

danboid, 2024-08-13:
Great to see this finally get fixed!

Like it or not Windows is one of the Big Three OS's, still the biggest by some accounts (certainly gaming) and I can't help but think that having to use Linux, in a VM or not, or WSL to build the repo and Uzebox dev tools might've put a few people off writing a Uzebox game? That excuse is no more!

closed 2024-08-13 by danboid