Uzebox/uzebox #137: Download page issues take 2

danboid, 2024-08-09:
Ooops!

I'm sorry but I had closed the previous ticket too soon.

The "Uzebox enclosure version 2" link links to v1.0 of the case.

Also, you have spelled "ensure" (insure) wrong in the comment "Check both Github repos to insure the one you need. "

How about "Check the README in the directory of the enclosures to ensure you are using the correct one."?

"Check both Github repos" doesn't really make sense as they're both in the same repo, just in different dirs.

uze6666, 2024-08-09:
Fixed the link and made the message clear enough I think. I mean, the link brings them right onto github with a big picture that show which connector to use. I can't make it more fool proof! So hope that's good enough to close the issue.

danboid, 2024-08-09:
Very nearly,now!

Could you please fix the spelling of readme.md (reame.md) on the Downloads page.

uze6666, 2024-08-12:
Done.

closed 2024-08-12 by danboid