Uzebox/uzebox #138: Add Uzenet module assembly guide PDF to repo

danboid, 2024-08-13:
I'm sure there was a Uzenet PDF assembly guide but it doesn't seem to have been uploaded to the repo?

It is almost hard to go wrong building it if you have bought the Uzenet module kit so it doesn't seem very worthwhile going to the trouble of rewriting it if @uze6666 doesn't still have a copy.

uze6666, 2024-08-15:
Can't find an assembly guide for the Uznet module. Perhaps you say the section in the portable assembly guide? 

Any any case, leave this issue open. Next time I assemble one, I'll snap some pics and make the guide.

danboid, 2024-08-16:
I thought you had sent one to me somehow but I failed to find it so maybe I imagined it?

I'm not sure a guide is required because I don't think anything can be inserted back to front.

closed 2024-08-16 by danboid