Uzebox/uzebox #144: Missing info and links in ESP programmer assembly guide

danboid, 2024-09-24:
I would like to build the ESP programming jig so I referred to the BOM in the jig assembly guide:

https://github.com/Uzebox/uzebox/blob/master/schematics/ESP-12F-Programming-Jig/V2.1/assembly-guide.md

The first bit of info I noticed that was absent is the pitch of the FTDI 90 degree header, which I presume is 2 mm?

We should also update the jig assembly guide to include links to at least one vendor for each item required.

It seems we can probably use DigiKey Part Number 17-2314878-8-ND

https://www.digikey.co.uk/en/products/detail/te-connectivity-amp-connectors/2314878-8/11502809

for the 8 position female header, 2mm pitch. Where did you buy your Uzenet friendly female headers from @uze6666, if it wasn't the one I've linked here.

danboid, 2024-09-24:
As for attaching a ESP 8266 to a Uzebox PCB using headers, this could be used:

CONN HEADER VERT 8POS 2MM - Digikey 952-2278-ND

https://www.digikey.co.uk/en/products/detail/harwin-inc/M22-2010805/3728242

danboid, 2024-09-24:
> It seems we can probably use DigiKey Part Number 17-2314878-8-ND
> 
> https://www.digikey.co.uk/en/products/detail/te-connectivity-amp-connectors/2314878-8/11502809
> 
> for the 8 position female header, 2mm pitch.

This costs twice as much as the one listed on the jig BOM so nope.

uze6666, 2024-09-26:
All the required parts are indeed listed in the BOM file. The 90 deg header is 2.54mm pitch and the 8 pins female one is 2mm. For nw I prefer to keep the bom separated, it's easier to quickly browse for the part numbers instead of going through the assembly guide, I have update the guide to refer to the BOM and added the 2.5mm mentioned on the 90deg header.

danboid, 2024-09-27:
Fair enough. I'm going to close this now.

closed 2024-09-27 by danboid