Uzebox/uzebox #150: UART MIDI IN support for MusicDemo

danboid, 2025-02-04:
UART MIDI input doesn't work for the MusicDemo version included in this repo but Uze posted an update to MusicDemo with working UART MIDI input support on the Uzebox forums here:

https://uzebox.org/forums/viewtopic.php?p=33750#p33750

It would be good to see this fixed code upstreamed.

I have attached a packrom .uze build of the UART MIDI capable UzeSynth / MusicDemo.uze 

[MusicDemo+UART-MIDI.zip](https://github.com/user-attachments/files/18659063/MusicDemo%2BUART-MIDI.zip)

weber21w, 2025-02-15:
I don't see any problem, and I think you are the only person to successfully use that demo in a long time. In theory the new code should be better than the old due to HSYNC UART implementation not available back then.

If @uze6666 has nothing against this, I say we upload his new version? I don't think there is anything specific about the new version which would preclude using the original PCB version with an actual MIDI plug and optical isolator?

danboid, 2025-02-16:
I have uploaded a YT video to demonstrate this Uzesynth feature:

```
There is an experimental build of Uzesynth aka MusicDemo available on the Uzebox forums that lets you use the Uzebox console as a MIDI controlled chiptune synthesizer via the Uzebox UART headers. It's not really much use a MIDI synth yet as you would have to recompile Uzesynth every time you wanted to change a patch but hopefully someone will create a more fully featured MIDI controlled synth for the Uzebox.

I didn't go to the effort of creating custom patches for this MIDI playback demonstration so it doesn't sound much like the actual game music but you can still make the tune out just using the default Uzesynth patches.

For more info see [https://uzebox.org/wiki/UART#UART_MIDI
```

https://www.youtube.com/watch?v=rwgfueckheI