Uzebox/uzebox #166: Build the qe text editor for UzeboxPC and add it to one of the default disk images

danboid, 2025-09-17:
One of the CP/M disk images included in `UzeboxPC-V1.zip` includes `ED.COM` which I presume is a text editor but I haven't been able to get it to run successfully, I only get the error

```
DISK OR DIRECTORY FULL
```

When I run `ed` from the UzeboxPC prompt.

I usually use vim when editing text in a (Linux/UNIX/macOS) terminal so I would quite like something similar for UzeboxPC. It turns out that the ZX Spectrum Next OS, NextZXOS, includes a vim alike editor that was borrowed from cpmish, qe:

https://github.com/davidgiven/cpmish/blob/master/cpmtools/qe.c

Wouldn't it be cool if both UzeboxPC and NextZXOS included qe as standard?

uze6666, 2025-10-01:
I spent over 12 hours trying to compile it on Windows without success. Seem to either missing files or they are wrongly named. Runnning under MSYS2 UCRT64 fails with all kind of errors but MSYS2 MSYS shell work. But then running ack just give "ack: Cannot find description for fe".  Where you able to build it under Linux or Windows/Msys?

danboid, 2025-10-01:
I built ack but not qe and the dev hasn't explained how to build cpmish or qe.

danboid, 2025-10-01:
Unfortunately there doesn't seem to be a Brother PowerNote emulator we could use to install CP/Mish and extract qe from. If we could work out how to extract the files from the .APL files, that might help.

```
root@SALP10185-UB:/home/dan/Downloads# mount -o loop pn8510.img /mnt2
root@SALP10185-UB:/home/dan/Downloads# cd /mnt2/
root@SALP10185-UB:/mnt2# file CPMISH.APL 
CPMISH.APL: data
root@SALP10185-UB:/mnt2# strings CPMISH.APL 
9:>p
```